efabless/caravel_board #12: config builder doesn't work as expected

mattvenn, 2022-12-23:
when I run Make, it will build the required gpio_config_data.c, but not from the local gpio_config_[io|def].py

https://github.com/efabless/caravel_board/blob/main/firmware_vex/matt_test/Makefile#L19

that's because the config builder includes local example python libs

https://github.com/efabless/caravel_board/blob/732b0792fee67d45fe970703807b90e371d25af9/firmware_vex/gpio_config/gpio_config_builder.py#L16

I fixed this by modifying the builder to include the real files, and then copied it into the firmware directory so that it picks up the local files.

I suggest adding command line arguments to explicitly pass the configs to read, and then the makefile can call it from the original position, but use local files. The local files could also be automatically passed to the builder through make rules.

jeffdi, 2023-02-08:
Can you please confirm if this is still an issue using the latest version with the gpio_test example?

mattvenn, 2023-02-09:
yes it's been solved

closed 2023-02-09 by mattvenn