efabless/caravel_board #16: Winbond SRAM not found

ayushman22git, 2023-01-12:
Hi, 

I am getting the following error when I run the IO diagnostic issue. 
[image]
 
Does anyone have any idea why this is the case and how to go about solving it

Thanks 
Ayushman

proppy, 2023-01-18:
I'm also hitting this issue.

proppy, 2023-01-18:
enabling debug for `erase` give a bit more information:
https://github.com/efabless/caravel_board/commit/f2a1839466fe9074e0ea3c444c1ab04057b13fce

```
(env) nucleo 🍊 make run
mpremote connect /dev/ttyACM0 exec "import machine; machine.reset()"
 
===================================================================
== Beginning IO configuration test.  Testing LOW IO chain...     ==
===================================================================
 

*** flashing Caravel
 
Caravel data:
   mfg        = 0456
   product    = 11
   project ID = 00000000
 
Resetting Flash...
status = 0xff
 
JEDEC = b'ffffff'
Winbond SRAM not found
```

proppy, 2023-01-18:
Getting a little bit further after fitting a different chip:
```
(env) nucleo 🍧 make run
mpremote connect /dev/ttyACM0 exec "import machine; machine.reset()"
 
===================================================================
== Beginning IO configuration test.  Testing LOW IO chain...     ==
===================================================================
 

*** flashing Caravel
 
Caravel data:
   mfg        = 0456
   product    = 11
   project ID = 00000000
 
Resetting Flash...
status = 0x00
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x0
Winbond SRAM not found
```

proppy, 2023-01-18:
Got a little bit further after discarding an intermediate usb hub and adding extra delay: https://github.com/proppy/caravel_board/commit/ec81d07497b41ddc857ebae486d4a7950e3953a7
```
>>> import io_config
>>> io_config.run()
 
===================================================================
== Beginning IO configuration test.  Testing LOW IO chain...     ==
===================================================================
 

*** flashing Caravel
 
Caravel data:
   mfg        = 0456
   product    = 11
   project ID = 00000000
 
Resetting Flash...
status = 0x00
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x0
 
Caravel data:
   mfg        = 0456
   product    = 11
   project ID = 00000000
 
Resetting Flash...
 
JEDEC = b'ef4016'
setting address to 00000000
addr 0x0: flash page write successful (1)
addr 0x100: flash page write successful (1)
addr 0x200: flash page write successful (1)
addr 0x300: flash page write successful (1)
addr 0x400: flash page write successful (1)
addr 0x500: flash page write successful (1)
addr 0x600: flash page write successful (1)
addr 0x700: flash page write successful (1)
addr 0x800: flash page write successful (1)
addr 0x900: flash page write successful (1)
addr 0xa00: flash page write successful (1)
addr 0xb00: flash page write successful (1)
setting address to 00000c00

total_bytes = 3072
^^^^^^^^ verifying  ^^^^^^^^
setting address to 00000000
addr 0x0: *** read compare FAILED ***
b'6f00000b13000000130000001300000013000000130000001300000013000000232e11fe232c51fe232a61fe232871fe2326a1fe2324b1fe2322c1fe2320d1fe232ee1fc232cf1fc232a01fd232811fd2326c1fd2324d1fd2322e1fd2320f1fd130101fcef0000118320c1038322810303234103832301030325c1028325810203264102832601020327c101832781010328410183280101032ec100832e8100032f4100832f0100130101047300203013010040170500001305c5f673105530130500009305800f17160000130686b3630cb500832606002320d50013054500130646006ff0dffe13050010930580106308b50023200500130545006ff05fff'
<----->
b'ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
addr 0x100: *** read compare FAILED ***
b'371500001305058873104530ef00d01c6f000000130101ff2326810013040101130000000324c1001301010167800000130101ff2326810013040101130000000324c1001301010167800000130101fe232e8100130401022326a4fe8327c4fe739007bc130000000324c1011301010267800000130101ff23261100232481001304010113050000eff05ffcb75700f09387c70123a00700b75700f09387c7001307100023a0e700b75700f09387070183a7070063980700130710002310e0106f008001b75700f09387c7011307100023a0e700130000008320c100032481001301010167800000130101ff2326810013040101130000000324c10013010101'
<----->
b'ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
addr 0x200: read compare successful
addr 0x300: *** read compare FAILED ***
b'13850735eff0dff1b70700261307e00023a0e700b7c7000013850735eff05ff0b70700261307600023a0e700b7c7000013850735eff0dfee130000008320c100032481001301010167800000130101fe232e1100232c810013040102232604fe6f000004b70700261307600023a0e700b7c7000013850735eff09feab70700261307600123a0e700b7c7000013850735eff01fe98327c4fe938717002326f4fe0327c4fe9307900fe3dee7faeff05ff3130000008320c101032481011301010267800000130101fe232e1100232c810013040102930710002326f4fe9307000083c707002324f4feeff05ff66f00c005130700008327c4feb307f70003c70700'
<----->
b'ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
addr 0x400: read compare successful
addr 0x500: read compare successful
addr 0x600: *** read compare FAILED ***
b'032481011301010267800000130101fd232611022324810213040103232ea4fc232604fe6f004001eff05fec8327c4fe938717002326f4fe0327c4fd8327c4fee354f7feb72726001385075aeff01fe2130000008320c102032481021301010367800000130101fd232611022324810213040103232ea4fc232604fe6f004001eff0dfef8327c4fe938717002326f4fe0327c4fd8327c4fee354f7feb72726001385075aeff09fdc130000008320c102032481021301010367800000130101fe232e810013040102b70700261307600623a0e700b70700261307600723a0e700b70700261307600023a0e700b70700261307600123a0e700232604fe6f008002'
<----->
[... cut ...] fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
setting address to 00000c00

total_bytes = 3072
 
Caravel data:
   mfg        = 0456
   product    = 11
   project ID = 00000000
 
Resetting Flash...
 
JEDEC = b'ffffff'
Winbond SRAM not found
>>> 
```

yesilzeytin, 2023-01-18:
I got the same issue. First, the green LED flashed 5 times initially as expected. Then, after red LED1 and blue LED2 alternate a few times, the red LED3 lights on and stands still with no flashing, and the green LED lighting still as well. So, the SRAM issue seems to be causing the test to not finish.

proppy, 2023-01-18:
@mattvenn just demonstrated he was able to get past this error using a [usb-powered hub](https://capablerobot.com/products/programmable-usb-hub/) with a good quality usb cable.

The power meter on the usb hub showed as much as 290mA during gpio characterization which is really close to what's documented in the nucleo board datasheet (as discussed in #23)

proppy, 2023-01-19:
I managed to get past the error with a combination of a better usb cable + usb powered hub.

jeffdi, 2023-02-08:
Fixed in v1.2.0

closed 2023-02-08 by jeffdi

mguthaus, 2023-02-08:
For posterity, what was the fix in 1.2.0?