efabless/caravel_board #17: Makefile should not mess up with the $PATH

proppy, 2023-01-12:
This currently break integration with currently activated venv or conda environment:
```
$ python3 -m venv env
$ source env/bin/activate
$ cd firmware_vex/nucleo/
$ make run
mpremote connect /dev/cu.usbmodem34203 exec "import machine; machine.reset()"
/bin/sh: 1: mpremote: not found
```

If you remove the line at  https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/Makefile#L2 it works as expected (`mpremote` get resolve from the currently activated environment).

ayushman22git, 2023-01-13:
This thread should be helpful 

[https://open-source-silicon.slack.com/archives/C022XAPHD9C/p1673043313859249]

proppy, 2023-01-18:
https://github.com/proppy/caravel_board/commit/ce91da71f58b7d456957d71ee5c45a71f3bb7d4c was enough for me for  the Makefile to pick up things in my local venv.