efabless/caravel_board #18: path to the usb device should not be hardcoded

proppy, 2023-01-12:
currently path the usb device is hardcoded in https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/Makefile#L5 which seems to use a path specific to a macosx installation.

On Linux based operating system, the path is likely to be different (`/dev/ttyACM0`, `/dev/ttyUSB0`) so something that detect the nucleo board is probably desirable:

proppy, 2023-01-18:
That's also true of the mount path in https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/Makefile#L9.

As a reference here is what I change to in order to reference the right devices on GNU/Linux:
https://github.com/proppy/caravel_board/commit/a113f39025f0b480152abd282bf8da47106492d5

proppy, 2023-01-18:
duplicate of #10?

jeffdi, 2023-02-08:
This has been updated to automatically set the path (v1.2.0).

closed 2023-02-08 by jeffdi