efabless/caravel_board #19: `make run` should provide informative log messages

proppy, 2023-01-12:
Thje current diagnosis is simply running `machine.reset()`:
https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/Makefile#L64
and doesn't produce any output.
```
(env) nucleo 🍡 make run
mpremote connect /dev/ttyACM0 exec "import machine; machine.reset()"
MicroPython v1.19.1-616-g5987130af-dirty on 2022-11-20; NUCLEO-F746ZG with STM32F746
Type "help()" for more information.
>>> 
```
and display a led combination that's not documented in the accompanying paper note (twe red leds, no blinking):
[image]

proppy, 2023-01-13:
Looks like my board was not preflashed:
```
>>> import os
>>> os.listdir()
['boot.py', 'main.py', 'pybcdc.inf', 'README.txt']
>>> open('boot.py').read()
"# boot.py -- run on boot to configure USB and filesystem\r\n# Put app code in main.py\r\n\r\nimport machine\r\nimport pyb\r\npyb.country('US') # ISO 3166-1 Alpha-2 code, eg US, GB, DE, AU\r\n#pyb.main('main.py') # main script to run after this one\r\n#pyb.usb_mode('VCP+MSC') # act as a serial and a storage device\r\n#pyb.usb_mode('VCP+HID') # act as a serial device and a mouse\r\n"
>>> open('main.py').read()
'# main.py -- put your code here!\r\n'
>>> open('README.txt').read()
"This is a MicroPython board\r\n\r\nYou can get started right away by writing your Python code in 'main.py'.\r\n\r\nFor a serial prompt:\r\n - Windows: you need to go to 'Device manager', right click on the unknown device,\r\n   then update the driver software, using the 'pybcdc.inf' file found on this drive.\r\n   Then use a terminal program like Hyperterminal or putty.\r\n - Mac OS X: use the command: screen /dev/tty.usbmodem*\r\n - Linux: use the command: screen /dev/ttyACM0\r\n\r\nPlease visit http://micropython.org/help/ for further help.\r\n"
>>> 
```
Does that mean I need to run `make copy` before running `make run`?

proppy, 2023-01-13:
running `make copy` seems to hang halfway:
```
(env) nucleo 💀 make copy
mpy-cross flash.py
mpy-cross io_config.py
mpy-cross gpio_config_builder.py
mpy-cross i2c.py
mpy-cross nucleo_api.py
mpy-cross boot.py
mpremote connect /dev/ttyACM0 cp flash.mpy i2c.mpy gpio_config_builder.mpy nucleo_api.mpy gpio_config_def.mpy :
cp flash.mpy :
 ... copying  11% [##------------------]
```
is this https://github.com/micropython/micropython/issues/8386 ?

proppy, 2023-01-13:
I'm having more reliable result with https://github.com/dhylands/rshell for the copy operations.

proppy, 2023-01-18:
The current flash sequence `make run` also make it very hard to find out where the failure are located:
- no stacktrace
- no context
- very similar log error message in https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/flash.py

jeffdi, 2023-02-08:
your board was not preflashed like other user boards were.  the instructions for reflashing has been updated.

closed 2023-02-08 by jeffdi