efabless/caravel_board #20: ValueError: incompatible .mpy file

ayushman22git, 2023-01-16:
Getting the following error when do make run. 
ayushman_ubuntu@UbuntuLTS:~/caravel_board/firmware_vex/nucleo$ make run
mpremote connect /dev/ttyACM0  exec "import machine; machine.reset()"
Traceback (most recent call last):
  File "main.py", line 2, in <module>
  File "io_config.py", line 1, in <module>
  File "nucleo_api.py", line 3, in <module>
ValueError: incompatible .mpy file
MicroPython v1.19.1-616-g5987130af-dirty on 2022-11-20; NUCLEO-F746ZG with STM32F746

Type "help()" for more information.

@msaligane

proppy, 2023-01-17:
Maybe an outdated version of the mpy files on the device?

Did you try:
```
mpremote ls
```

ayushman22git, 2023-01-18:
> 
This is the response I get with mpremote ls

[image]

proppy, 2023-01-18:
@ayushman22git can you give more details on how you connected the board to the host computer?

proppy, 2023-01-18:
@ayushman22git did you also run:
```
make compile
make copy
```

ayushman22git, 2023-01-18:
@proppy I connected using the CN1 connector on Nucleo to my USB host. Utilized mpremote. I had not run these 2 commands. However I have now run these commands. the following is the output 
[image]
Copy is not proceeding beyond this point. However, after several iterations the copy operation halted as in figure 2. 
[image]
Not proceeding  beyond this point.  When interrupt this at times I see mostly the following output. It seems to hang. In some trials, see earlier error message with make run as well. 
[image]

proppy, 2023-01-18:
another possible workaround is not not compile mpy file and simply copy py file around:
https://github.com/proppy/caravel_board/commit/d19877f864fbc368b3874f8979909bdc99b4f162

proppy, 2023-01-18:
@ayushman22git one gotcha that @mattvenn figured out during his recent demonstration, is that a more reliable copy is possible by:
- reflashing the board with `make F746ZG`
- connecting another usb cable on the opposite side of the st-link connector
- mounting the corresponding volume that's showing up
- updating the path to the volume in https://github.com/efabless/caravel_board/blob/main/firmware_vex/nucleo/Makefile#L9
- copying the file using `make copy2`
-

jeffdi, 2023-02-08:
proppry instrustuctions are correct and are now included to the README

closed 2023-02-08 by jeffdi