efabless/caravel_board #37: risc-v toolchain requirement should be documented

proppy, 2023-01-30:
When using the risc-v toolchain from debian:
```
ii  gcc-riscv64-unknown-elf                              12.1.0-7+11                                             amd64        GCC cross compiler for Risc-V processors
```
I get the following error when trying to compile some of the test firmware of the repo:
```
riscv64-unknown-elf-gcc -I/home/proppy/src/github.com/efabless/caravel_board/firmware_vex/gpio_test -I../ -I../generated/ -O0 -mabi=ilp32 -march=rv32i -D__vexriscv__ -Wl,-Bstatic,-T,../sections.lds,--strip-debug -ffreestanding -nostdlib -o gpio_test.elf ../crt0_vex.S ../isr.c gpio_test.c
../crt0_vex.S: Assembler messages:
../crt0_vex.S:59: Error: unrecognized opcode `csrw mtvec,a0', extension `zicsr' required
../crt0_vex.S:87: Error: unrecognized opcode `csrw mie,a0', extension `zicsr' required
../irq_vex.h: Assembler messages:
../irq_vex.h:31: Error: unrecognized opcode `csrw 3008,a5', extension `zicsr' required
make: *** [Makefile:30: gpio_test.elf] Error 1
```

It would be nice to document the toolchain requirements and guide the user thru the required setup.

proppy, 2023-02-16:
Found pre-compiled version of the risc-v toolchain that included the `zicsr` extension here:
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack

kavyasreedhar, 2023-03-01:
For MacOS Ventura, this worked for me:

```
brew tap riscv-software-src/riscv
brew install riscv-tools
```

On the `brew install riscv-tools` step, I ran into this error: `Error: Your Command Line Tools are too outdated`
Following the steps here https://coady.tech/macos-command-line-tools-outdated/ fixed that issue and then I could rerun `brew install riscv-tools`

britovski, 2023-03-31:
For CentOS or rh users, can use the riscv_install scripts here: https://github.com/britovski/micro_skel/tree/main/scripts

appujee, 2023-04-03:
i ran into this issue as well. the default toolchain installed via `sudo apt install gcc-riscv64-unknown-elf` gives this error.

PR #86 (2023-08-01): Updated compiler target architecture
Modified the compiler architecture flag from `-march=rv32i` to `-march=rv32i_zicsr` to work with modern GCC. This should resolve #37 and #87
closed 2023-08-01 by jeffdi