efabless/caravel_board #63: UART function definitions are missing

abdullahyildiz, 2023-02-14:
It seems that definitions of the function declarations in [uart.h](https://github.com/efabless/caravel_board/blob/b4da7033cfa695726a30b435dcb263405106cca3/firmware_vex/uart.h) are missing in [firmware_vex](https://github.com/efabless/caravel_board/tree/main/firmware_vex):

```C
#ifndef __UART_H
#define __UART_H

#ifdef __cplusplus
extern "C" {
#endif

#define UART_EV_TX	0x1
#define UART_EV_RX	0x2

void uart_init(void);
void uart_isr(void);
void uart_sync(void);

void uart_write(char c);
char uart_read(void);
int uart_read_nonblock(void);

#ifdef __cplusplus
}
#endif

#endif
```

abdullahyildiz, 2023-03-08:
There is also no register defined for setting the UART baud rate (i.e., divisor) in [csr.h](https://github.com/efabless/caravel_board/blob/main/firmware_vex/csr.h). Is it included as a register field in one of the other CSRs?

proppy, 2023-03-15:
@thesourcerer8 commented about getting uart to work here: 
https://github.com/thesourcerer8/mpw2-stdcell-tests/blob/main/uart_test/uart_test.c

proppy, 2023-03-15:
but it would be nice to also have those defined in the nucleo micropython `machine` environment, currently it returns:
```
uart = UART(0, 9600)                         # init with given baudrate

from machine import UART
​
uart = UART(0, 9600)                         # init with given baudrate
uart.init(9600, bits=8, parity=None, stop=1) # init with given parameters
uart.read()

Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
ValueError: UART(0) doesn't exist
```

proppy, 2023-03-16:
it particular it seems that `UART8` (which correspond to https://caravel-harness.readthedocs.io/en/latest/uart.html) is not plugged into the default nucleo HAL:
https://github.com/micropython/micropython/blob/05bb26010e4a466a82cfed179f8d8d0b406a78ca/ports/stm32/boards/NUCLEO_F746ZG/mpconfigboard.h#L32-L39

abdullahyildiz, 2023-03-16:
@proppy,

As far as I know, MicroPython is a software stack which runs on Nucleo board and UART peripheral on Caravel management SoC is not controlled by MicroPython.

proppy, 2023-03-16:
@abdullahyildiz yep, but the hardware pin that export the hardware bus are physically connected to the Nucleo board, so it should be possible to surface them to the micropython environment thru the HAL.

abdullahyildiz, 2023-03-16:
@proppy,

UART peripheral on management SoC is not connected to Nucleo board. We were able to interact with it by standard UART cable.

proppy, 2023-03-16:
@abdullahyildiz I think they are electrically connected, UART is on `E7` and `F7`, respectively mapped to `mprj_io[5]` and `mprj_io[6]` which onnect to `CN11` `PF0` and `CN12` `PE11` of the nucleo Morpho connector:
[image]
but don't seem to be currently surfaced in the micropython HAL UART definitions.

proppy, 2023-03-17:
found out that we can actually bridge one of the existing UART port defined here:
https://github.com/micropython/micropython/blob/master/ports/stm32/boards/NUCLEO_F746ZG/mpconfigboard.h#L38-L39

by setting the pin `#9` to `C_DISABLE` in `gpio_config_io.py` and leaving its register untouched in `set_registers` and then physically bridging `IO[9]` (which correspond to `MICROPY_HW_UART6_RX` (`G9`) with `IO[6]` using a jumper cable:
[image]

The data from the caravel serial port can then be read from the micropython environment using:
```
uart = machine.UART(6, 9600)
print(uart.read())
```