efabless/caravel_board #69: caravel_Nucleo percurment

CarlFK, 2023-04-04:
I'd like a few of these boards (less than 10), I'm assuming I have to get them manufactured and assembled myself.  
What company was used?

Is there some way to reference the previous run so I can ask for "more of that" ?

I did not have a chip made, so I'm not eligible for anything that participants are, but I was given some of the chips to experiment with.

CarlFK, 2023-04-06:
also, it seems the readme is now out of date given  #68 

also also, what is the difference between development and Nucleo?

urish, 2023-06-24:
You may be able to obtain some through [TinyTapeout](https://TinyTapeout.com)

Aidan-McNay, 2023-07-20:
If you already have the chips, the KiCad PCB files are in the `hardware` directory (for whichever version of the board you're looking for). These should hopefully plug in without any change with any Caravel chips