efabless/caravel_board #82: read compare error when flashing blink firmware for GF-MPW-0

proppy, 2023-06-29:
Getting the following error when trying to follow GF-MPW-0 instructions:
https://github.com/efabless/caravel_board/tree/main/gf180

```
python3 ../util/caravel_hkflash.py blink.hex
Success: Found one matching FTDI device at ftdi://ftdi:232h:2:8/1
 
Caravel data:
   mfg        = 0456
   product    = 20
   project ID = 0c002b63
 
Resetting Flash...
status = 0xe0
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x2
setting address to 0x0
addr 0x0: flash page write successful
addr 0x100: flash page write successful
addr 0x200: flash page write successful

total_bytes = 544
status reg_1 = 0x16
status reg_2 = 0x10
************************************
verifying...
************************************
status reg_1 = 0x6
status reg_2 = 0x1
setting address to 0x0
addr 0x0: *** read compare FAILED ***
b'930000009301000013020000930200001303000093030000130400009304000013050000930500001306000093060000130700009307000013080000930800001309000093090000130a0000930a0000130b0000930b0000130c0000930c0000130d0000930d0000130e0000930e0000130f0000930f0000170500001305851a970500f1938505f8170600f1130686f763dcc5008326050023a0d5001305450093854500e3c8c5fe170500f1130585f5970500f1938505f56358b5002320050013054500e34cb5feef00400b6f000000b70200281303001223906200a381020063080602130f80009373f60f93de73002380d20193ee0e012380d20193931300'
<----->
b'930d0000130e0000930e0000130f0000930f0000170500001305851a970500f1938505f8170600f1130686f763dcc5008326050023a0d5001305450093854500e3c8c5fe170500f1130585f5970500f1938505f56358b5002320050013054500e34cb5feef00400b6f000000b70200281303001223906200a381020063080602130f80009373f60f93de73002380d20193ee0e012380d20193931300ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
addr 0x100: *** read compare FAILED ***
b'93f3f30f130fffffe3120ffe2380620063800506130f000283230500930f800093def3012380d20193ee0e012380d20183ce020093fe2e0093de1e0093931300b3e3d301130fffff63180f00232075001305450083230500938fffffe3920ffc9385f5ffe39c05fa63040f002320750013030008a381620067800000130101fe232e810013040102930710002324f4feb70700219387470023a00700b70700219387870023a00700b70700219387c70023a00700b70700211307100023a0e700b707002123a00700232604fe6f0000018327c4fe938717002326f4fe0327c4feb7170000938777bbe3d4e7feb70700211307100023a0e700232604fe6f000001'
<----->
b'98400004984000009848000498480000985000049850000098580004985800009860000498600000986800049868000098700004987000009878000498780000b8280000982c28d4b828078c9c282fc0b8300788983437bb1ee62804193028011d06a800982a28049c2a28071e462ff0b8280788982c2facb828078c9c282fab1ac5a80119002800982a28071a65aff77802005b78000005b8100140981800911c8310051c08100318403010987c00049b9fb07c9ef398011c06900c9f7070091c06900c9c989807ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff'
addr 0x200: *** read compare FAILED ***
b'8327c4fe938717002326f4fe0327c4feb7170000938777bbe3d4e7fe6ff05ffa'
<----->
b'61400012614000026160001261600002618000126180000261a0001261a00002'

total_bytes = 544
pll_trim = b'63'
```

is there a way to alter the baud rate for the flash operation to be more reliable?

proppy, 2023-06-29:
tried to remove the usb hub, and with a different usb cable without success.

ThorKn, 2023-06-29:
After erasing the chip i get the status 0x0. I don't know what that status means, but its different from yours. And before writing to the flash:

python3 ../util/caravel_hkflash.py blink.hex
Success: Found one matching FTDI device at ftdi://ftdi:232h:5:14/1
 
Caravel data:
   mfg        = 0456
   product    = 20
   project ID = 18004515
 
Resetting Flash...
status = 0x00
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x0
setting address to 0x0
addr 0x0: flash page write successful
addr 0x100: flash page write successful
addr 0x200: flash page write successful
addr 0x300: flash page write successful
addr 0x400: flash page write successful
addr 0x500: flash page write successful
addr 0x600: flash page write successful
addr 0x700: flash page write successful
addr 0x800: flash page write successful
addr 0x900: flash page write successful
addr 0xa00: flash page write successful
addr 0xb00: flash page write successful
addr 0xc00: flash page write successful
addr 0xd00: flash page write successful
addr 0xe00: flash page write successful
setting address to 0xe2c
addr 0xe2c: flash page write successful

total_bytes = 3728
status reg_1 = 0x0
status reg_2 = 0x2
************************************
verifying...
************************************
status reg_1 = 0x0
status reg_2 = 0x2
setting address to 0x0
addr 0x0: read compare successful
addr 0x100: read compare successful
addr 0x200: read compare successful
addr 0x300: read compare successful
addr 0x400: read compare successful
addr 0x500: read compare successful
addr 0x600: read compare successful
addr 0x700: read compare successful
addr 0x800: read compare successful
addr 0x900: read compare successful
addr 0xa00: read compare successful
addr 0xb00: read compare successful
addr 0xc00: read compare successful
addr 0xd00: read compare successful
addr 0xe00: read compare successful
setting address to 0xe2c
addr 0xe2c: read compare successful

total_bytes = 3728
pll_trim = b'a8'

proppy, 2023-07-03:
Building the firmware from `gf180/blink` does fix the issue, thanks @asinghani!
```
# python3 ../util/caravel_hkflash.py blink.hex
Success: Found one matching FTDI device at ftdi://ftdi:232h:2:8/1
 
Caravel data:
   mfg        = 0456
   product    = 20
   project ID = 180056c6
 
Resetting Flash...
status = 0x00
 
JEDEC = b'ef4016'
Erasing chip...
done
status = 0x0
setting address to 0x0
addr 0x0: flash page write successful
addr 0x100: flash page write successful
addr 0x200: flash page write successful
addr 0x300: flash page write successful
addr 0x400: flash page write successful
addr 0x500: flash page write successful
addr 0x600: flash page write successful

total_bytes = 1660
status reg_1 = 0x0
status reg_2 = 0x2
************************************
verifying...
************************************
status reg_1 = 0x0
status reg_2 = 0x2
setting address to 0x0
addr 0x0: read compare successful
addr 0x100: read compare successful
addr 0x200: read compare successful
addr 0x300: read compare successful
addr 0x400: read compare successful
addr 0x500: read compare successful
addr 0x600: read compare successful

total_bytes = 1660
pll_trim = b'63'

PR #84 (2023-08-01): gf180/README: fix makefile path
Fixes #82
closed 2023-08-01 by jeffdi