efabless/caravel_board #87: Recent versions of RISC-V gcc require "zicsr" in the compile line.

RTimothyEdwards, 2023-07-21:
Recently when I compile the RISC-V toolchain, I cannot get caravel_board examples to compile unless I modify the compile line to change `-march=rv32i` to `-march=rv32i_zicsr`.

Apparently this is because there were optional libraries previously included into the code by default that no longer are;  the "zicsr" library is one of these.  I believe that means that making the change is backwardly-compatible with earlier versions of the RISC-V gcc toolchain, but that needs to be confirmed.

This problem has been reported by multiple people.

Aidan-McNay, 2023-07-22:
@jeffdi I noticed this as well - the PR #86 should resolve this :)

RTimothyEdwards, 2023-07-23:
@Aidan-McNay :  Thanks for pointing to your pull request.  I'll review it so that it can get merged.  (Heads up, @jeffdi )